tscircuit/schematic-trace-solver #29: New Phase To combine same-net traces that are close together

seveibar, 2025-08-29:
I think it's best to implement as a new pipeline phase to split up the logic

/bounty $100

[image]

seveibar, 2025-08-29:
To do this, you will need to find a reproduction, i'm not sure it's possible to create this state so please make sure to create a reproduction first to verify it's possible

Exceluyi, 2025-11-08:
/attempt #29

giajoe24, 2026-01-16:
/attempt #29

wazder, 2026-03-05:
/attempt #107 
Hi! I'd like to work on this bounty. I have experience with TypeScript and schematic layout pipeline design. My plan is to implement a new pipeline phase in schematic-trace-solver that combines same-net trace segments that are spatially close together. This phase will run after the initial routing phase, detect near-adjacent collinear segments on the same net, and merge them into unified trace segments. I will implement it as a clean, isolated pipeline step with full tests and submit a PR.

zhaog100, 2026-03-25:
/attempt #29

I just completed #86 (same-net trace merging in core), so I'm deeply familiar with the schematic trace pipeline. Plan:

1. Create a reproduction test case showing adjacent same-net traces not being merged
2. Implement a new pipeline phase `TraceSegmentMergerSolver` that:
   - Identifies same-net trace segments that are close together and aligned
   - Merges collinear segments into continuous wires
3. Add tests to verify the merge behavior
4. Submit PR with /claim #29

mitchellecm7, 2026-04-06:
/claim #29

https://github.com/user-attachments/assets/76f830c4-7219-41cc-8ddb-63968df587a4

- Before: iterations 0, traces separate
- After: iterations 4, traces combined and solved
- Solver reaches "Solved" state

This implements combining same-net trace segments that are close together. There's the demo video showing same-net trace segments being combined (iterations 0 → 4, solved).

The issue shows "rewarded" but I haven't received payment yet. Please review and confirm. Thanks!

selenaalpha77-sketch, 2026-04-25:
/attempt #29

I've submitted a PR implementing this feature. Please see my pull request for the implementation details.

selenaalpha77-sketch, 2026-04-25:
Updated solution: PR #226 (SameNetTraceMergeSolver) is my latest implementation. It merges same-net trace segments that run close in parallel (within 0.5 units) into single segments, reducing visual clutter. All CI checks passing.

/claim #29

sayidilxs-web, 2026-05-05:
Hello @algora-io, I would like to work on this issue. I am planning to implement the new pipeline phase to combine same-net trace segments as suggested. Please assign this to me.

janishahn, 2026-05-10:
/attempt #29

Implementation plan:
- Add a post-cleanup SameNetTraceCombiningSolver phase.
- Group routed traces by globalConnNetId and snap only close overlapping same-net horizontal or vertical internal segments onto a shared run.
- Preserve pin endpoints, leave different nets unchanged, and reject snaps that would introduce a different-net trace intersection.
- Wire the combined traces into downstream net-label placement and example phases.
- Include focused Bun tests plus full-suite, build, and format verification.

Iineman2, 2026-05-11:
/attempt #29

I found a current-main gap where the pipeline still has no dedicated same-net combining phase. I am submitting a conservative post-cleanup phase that only snaps close overlapping internal same-net horizontal/vertical segments onto a shared run, keeps pin endpoint segments fixed, and rejects snaps that would intersect different-net traces. Verification includes focused solver tests, example29/example30 snapshots, build, and changed-file formatting.

dclawd88, 2026-05-11:
/attempt #29

Working on a same-net trace consolidation phase that snaps close overlapping same-net segments while avoiding different-net intersections.

chenchong0215, 2026-05-11:
/claim #29

2balmprune, 2026-05-11:
/claim #29 https://github.com/tscircuit/schematic-trace-solver/pull/297

PR #297 adds the same-net trace merge pipeline phase, keeps endpoint segments fixed, rejects different-net intersections, and includes focused same-net/different-net tests. Local verification is listed in the PR body.

ron1nrest, 2026-05-17:
> To do this, you will need to find a reproduction, make sure to create a reproduction first to verify it's possible

#355

WhiteEagle-12, 2026-05-17:
@Mira-Mjodheim I submitted the concrete implementation in #365.

The approach is intentionally conservative:

- Run after `TraceCleanupSolver`, when traces are already simplified into mostly axis-aligned paths.
- Group traces by `globalConnNetId`, so segments from different nets are never candidates.
- Only consider horizontal-with-horizontal or vertical-with-vertical internal segments that are close and have overlapping projection ranges.
- Move the weaker/internal segment onto the shared axis, preserving both endpoints so pin/contact locations do not drift.
- Reject a candidate if it would introduce a stricter crossing with a different-net segment.
- Iterate a bounded number of passes until no more same-net segments can be aligned.

The goal is to solve the visual duplicate/near-parallel same-net runs shown in #29 without merging traces across nets or changing connection endpoints. I added focused solver tests plus existing example coverage around examples 29, 30, and 34.