tscircuit/schematic-trace-solver #46: New phase to move trace segments "out of the way" of traces

seveibar, 2025-09-06:
Find the most aesthetic (minimal collisions) to move segments such that netlabels don't overlap traces.

Netlabels already test for trace overlap when placed, but there is often a fallback where the netlabel can't go anywhere else, that's where this logic is relevant

Please implement as a new render phase. I recommend looking at specific segments and doing generateElbowVariants to see variants of that segment that avoid the overlap

[image]

/bounty $100

shivankurchavan, 2025-09-17:
/attempt #46

saish9901, 2025-09-27:
/attempt #46

closed 2025-11-11 by 0hmX